RisingTechOSS-devices/official_devices #2: New maintainer for Xiaomi 11T Pro (vili)

0mar99, 2024-12-03:
## Device you want to apply for
```
Xiaomi 11T Pro (vili)
```

## Email address
```
omarag9099@gmail.com
```

## Device tree sources

```
Device trees:
https://github.com/AOSP-for-vili/device_xiaomi_vili/tree/rising
https://github.com/AOSP-for-vili/device_xiaomi_sm8350-common/tree/rising

Vendor trees:
https://github.com/AOSP-for-vili/vendor_xiaomi_vili
https://github.com/AOSP-for-vili/vendor_xiaomi_sm8350-common.git

Kernel:
https://github.com/AOSP-for-vili/android_kernel_xiaomi_sm8350
```

## Link to the platform where you have released your unofficial build
```
XDA:
https://xdaforums.com/t/rom-unofficial-15-risingos-xiaomi-11t-pro-vili.4705166/#post-89825055

My support group:
https://t.me/ViliSupport/509/24989

Official Xiaomi 11T Pro channel:
https://t.me/Xiaomi11TPro_Official/286702

```

## Other ROM's you maintain officially or unofficially
```
DerpFest (officially) and OrionOS (officially). 
Note: I strictly only maintain 2 official ROMs at a time, so I would be dropping support for OrionOS in favor of Rising
```

## Bugs on unofficial build
```
None
```

## How long have you been building for?
```
Almost 2 years. Since March, 2023
```

## SELinux status?
```
Enforcing
```

## Are you using a prebuilt kernel? If yes, why?
```
No.
```

## Contact
### Telegram username
```
@Omar99A
```

hipexscape, 2024-12-04:
Hi @0mar99,

Can you explain me how the following prop works ? Or does that even exists in AOSP ?

- https://github.com/AOSP-for-vili/device_xiaomi_sm8350-common/commit/d110c510c5f56a4c29c9fab0317251ad1952f0be

Regards,
@hipexscape,
On behalf of RisingOs team.

0mar99, 2024-12-04:
Hello. Yes, they do exist in AOSP. They are based on this commit found in DerpFest's and Orion's frameworks_base:
- https://github.com/DerpFest-AOSP/frameworks_base/commit/dbafc0400d27c2ec15232630d0d97f7d6ddebd86
- https://github.com/OrionOS-prjkt/android_frameworks_base/commit/8edc1498062ac65edf647ed374e83ac11bb7aa90

As the commit message implies, different fling velocities do provide an improvement to perceived responsiveness in fast scrolling cases, albeit very slight. Here's some google documentation regarding the idea of fling animations:
- https://developer.android.com/develop/ui/views/animations/fling-animation

As for why it is in my Rising branch, you are correct, it appears to be an oversight on my side at the time of adapting my DerpFest trees for RisingOS, as this change is not yet added on RisingOS source:
- https://github.com/RisingTechOSS/android_frameworks_base_new/commits/fifteen/core/java/android/view/ViewConfiguration.java

As such, I will remove it immediately. Thank you for pointing it out! 

I look forward to any other questions you may have

Best regards,
Omar

hipexscape, 2024-12-08:
> Hello. Yes, they do exist in AOSP. They are based on this commit found in DerpFest's and Orion's frameworks_base:
> 
> * [DerpFest-AOSP/frameworks_base@dbafc04](https://github.com/DerpFest-AOSP/frameworks_base/commit/dbafc0400d27c2ec15232630d0d97f7d6ddebd86)
> * [OrionOS-prjkt/android_frameworks_base@8edc149](https://github.com/OrionOS-prjkt/android_frameworks_base/commit/8edc1498062ac65edf647ed374e83ac11bb7aa90)
> 
> As the commit message implies, different fling velocities do provide an improvement to perceived responsiveness in fast scrolling cases, albeit very slight. Here's some google documentation regarding the idea of fling animations:
> 
> * https://developer.android.com/develop/ui/views/animations/fling-animation
> 
> As for why it is in my Rising branch, you are correct, it appears to be an oversight on my side at the time of adapting my DerpFest trees for RisingOS, as this change is not yet added on RisingOS source:
> 
> * https://github.com/RisingTechOSS/android_frameworks_base_new/commits/fifteen/core/java/android/view/ViewConfiguration.java
> 
> As such, I will remove it immediately. Thank you for pointing it out!
> 
> I look forward to any other questions you may have
> 
> Best regards, Omar

Kindly remove it

hipexscape, 2024-12-08:
Hi @0mar99,

I think this commit is authored by SebaUbuntu right ?

- https://github.com/AOSP-for-vili/device_xiaomi_sm8350-common/commit/e8f757adbc42b05d2c077d5fa4645751807eaa7c

Regards,
@hipexscape,
On behalf of RisingOs team.

0mar99, 2024-12-08:
You are right, I got it from Lineage's extract-utils repo. I've been using it for a long time, so I never thought about who to give authorship, since there's several edits done by different authors. Should I give authorship to SebaUbuntu in that case? Since he authored the original file

Reference:
- https://github.com/LineageOS/android_tools_extract-utils/commits/lineage-22.0/sort-blobs-list.py

RisenID, 2024-12-08:
Hey, please could you explain to me what the differences between villi and haydn are? Seems the hadyn trees have just been rebranded to villi.

RisenID, 2024-12-08:
In addition, please remove this and make sure your trees are compliant with the Code of Conduct as listed [Code of Conduct](https://github.com/RisingTechOSS-devices/official_devices/blob/main/conduct.md)
https://github.com/AOSP-for-vili/device_xiaomi_vili/commit/907653b4275bd054b81f998050f4141a7bf56693

0mar99, 2024-12-08:
> Hey, please could you explain to me what the differences between villi and haydn are? Seems the hadyn trees have just been rebranded to villi.

Hi! Kind of a short story, sorry. "Haydn" refers Xiaomi Mi 11i/11X Pro and Redmi K40 Pro, they're variants of the same device, which are part of the sm8350 platform. As for the rebranding to vili, about a year ago, the development was very stale. We were lucky to get help from a LineageOS developer named TheStrechh, who added unofficial support for vili to the latest LineageOS common trees and kernel at the time. Unfortunately he was not able to buy the device to support it officially, so instead of making the device tree from scratch, he decided to rebrand haydn's tree to vili, as they are very similar phones. 

As for your suggestions, I have removed KernelSU and any related changes from the kernel. Also I believe I've made sure my trees are compliant with the code of conduct, except for the branch names, as I'm thinking that's for when the repos are already uploaded in the Rising-devices org, but please correct me if I'm wrong.

hipexscape, 2024-12-15:
Any reasons for such long rebase ? I see then everyday

0mar99, 2024-12-15:
> Any reasons for such long rebase ? I see them everyday

Just fixing stuff as @RisenID mentioned and making sure my trees are compliant with your CoC. 
- https://github.com/RisingTechOSS-devices/official_devices/issues/2#issuecomment-2525443954

Also some restructuring as to not need any unnecessary external repos like Dolby Atmos, modified display HAL, etc. I am done with changes to my RisingOS trees since last week, as you can see in the history

hipexscape, 2024-12-18:
Looks fine from my end 
@manidweep @RisenID please check

manidweep, 2024-12-20:
its a no from my side, please apply again when you know the importance of authorship and credits where it is required

closed 2024-12-20 by hipexscape