RisingTechOSS-devices/official_devices #4: New maintainer for Ginkgo

Miracleprjkt, 2024-12-03:
## Device you want to apply for
```
Xiaomi Redmi Note 8 - Ginkgo
```

## Email address
```
Miraclev1@proton.me
```

## Device tree sources

```
https://github.com/Miracleprjkt/device_xiaomi_ginkgo
https://github.com/Miracleprjkt/android_kernel_xiaomi_ginkgo
https://github.com/Miracleprjkt/vendor_xiaomi_ginkgo
```

## Link to the platform where you have released your unofficial build
```
https://github.com/Miracleprjkt/Release
```

## Other ROM's you maintain officially or unofficially
```
Insert here
```

## Bugs on unofficial build
```
None
```

## How long have you been building for?
```
One Years
```

## SELinux status?
```
Enforcing
```

## Are you using a prebuilt kernel? If yes, why?
```
No.
```

## Contact
### Telegram username
```
@Miraclesupremacy
```

eun0115, 2024-12-04:
https://github.com/RisingOSS-devices/official/issues/78

you applied 4 days ago

manidweep, 2024-12-04:
> [RisingOSS-devices/official#78](https://github.com/RisingOSS-devices/official/issues/78)
> 
> you applied 4 days ago

he shifted it here since old org is depreciated

eun0115, 2024-12-04:
regarding with "Other ROM's you maintain officially or unofficially" and you put "insert here", i saw you blind building many times for other devices. it might be an issue with other core members, just a heads up on my side.

manidweep, 2024-12-04:
> regarding with "Other ROM's you maintain officially or unofficially" and you put "insert here", i saw you blind building many times for other devices. it might be an issue with other core members, just a heads up on my side.

yikes, if that is really the case
its a no from my side, if it is blind building

Miracleprjkt, 2024-12-04:
> regarding with "Other ROM's you maintain officially or unofficially" and you put "insert here", i saw you blind building many times for other devices. it might be an issue with other core members, just a heads up on my side.

Hmmm....sorry sir but i not blind build,i have the devices

RisenID, 2024-12-08:
Hey, we don't allow maintainers who maintain more than 3 roms, to maintain quality and support from that maintainer. Due to this reason, your application has been denied. Please feel free to apply in the future if the aforementioned circumstances change.

closed 2024-12-08 by RisenID

nayemghit, 2024-12-10:
I have been wanting to use a press room for a long time that can run all my favorite software smoothly, but with this Android version update, everyone is busy, so I installed this room. After using it for a few days, it seems like this is the kind of good room I was looking for. I have no words to say. I am going to thank the maintainer.