RisingTechOSS-devices/official_devices #5: New maintainer

Rakhshan7070, 2024-12-04:
## Device you want to apply for
```
Moto Edge 30 (Dubai)
```

## Email address
```
alirakhshan7070@gmail.com
```

## Device tree sources

```
https://github.com/Rakhshan7070/android_device_motorola_dubai
https://github.com/Rakhshan7070/android_device_motorola_sm7325-common
https://github.com/Rakhshan7070/proprietary_vendor_motorola_dubai
https://github.com/Rakhshan7070/proprietary_vendor_motorola_sm7325-common
https://github.com/Rakhshan7070/android_kernel_motorola_sm7325
```

## Link to the platform where you have released your unofficial build
```
https://sourceforge.net/projects/dubai-build-by-rakhshan/files/
```

## Other ROM's you maintain officially or unofficially
```
CrDroid
```

## Bugs on unofficial build
```
None
```

## How long have you been building for?
```
Months
```

## SELinux status?
```
Enforcing
```

## Are you using a prebuilt kernel? If yes, why?
```
No.
```

## Contact
### Telegram username
```
shan_rakh
```

RisenID, 2024-12-08:
Hey, it seems a lot of your builds have KSU built in. Could you explain why?

Rakhshan7070, 2024-12-08:
@RisenID 
Yes, most of my builds have KSU built in because most users prefer KSU & they requested me to do so.

manidweep, 2024-12-16:
approved from my end, need 1 more for confirmation

hipexscape, 2024-12-16:
Approved

hipexscape, 2024-12-16:
Hey @Rakhshan7070,

Kindly contact me on telegram

Username:- hipexscape

Regards,
Abhinav Kumar

closed 2024-12-16 by manidweep