RisingTechOSS-devices/official_devices #6: New maintainer for Realme 7/ Narzo 30/ Narzo 20 Pro (salaa)

ShitijHalder, 2024-12-04:
## Device you want to apply for
```
Realme 7/ Narzo 30/ Narzo 20 Pro (salaa)
```

## Email address
```
shitijhalder4@gmail.com
```

## Device tree sources

```
https://github.com/ShitijHalder/device_realme_salaa
https://github.com/ShitijHalder/kernel_realme_salaa
https://github.com/ShitijHalder/vendor_realme_salaa
```

## Link to the platform where you have released your unofficial build
```
[Telegram](https://t.me/shitijposts/332)
```

## Other ROM's you maintain officially or unofficially
```
Infinity X Officially. (Other rom if the people wants, once in a while only)
```

## Bugs on unofficial build
```
None.
```

## How long have you been building for?
```
Around 10 months for this device now.
```

## SELinux status?
```
Permissive.
```

## Are you using a prebuilt kernel? If yes, why?
```
No.
```

## Contact
### Telegram username
```
shitijnotop
```

eun0115, 2024-12-04:
explain why its permissive?

ShitijHalder, 2024-12-04:
> explain why its permissive?

Sorry, it's enforcing, I didn't see.

RisenID, 2024-12-05:
Hey thanks for the application. 
Due to the sheer number of rom bringup commits (I counted over 15) and someother random stuff like not cloning signing keys to vendor/lineage-priv like all roms allow for, i must reject your application.
Please feel free to apply in the future once you have more experience and when all the issues mentioned are fixed