RisingTechOSS-devices/official_devices #9: New maintainer

ZorEl212, 2024-12-09:
## Device you want to apply for
```
Redmi Note 8/8T (Ginkgo/Willow)
```

## Email address
```
yabsirad212@proton.me
```

## Device tree sources

```
https://github.com/ZorEl212/android_device_xiaomi_ginkgo-retrofit
https://github.com/ZorEl212/android_kernel_xiaomi_ginkgo
https://github.com/ZorEl212/proprietary_vendor_xiaomi_ginkgo
```

## Link to the platform where you have released your unofficial build
```
https://t.me/rogerbuilds/818
```

## Other ROM's you maintain officially or unofficially
```
ArrowOS Extended 
```

## Bugs on unofficial build
```
None
```

## How long have you been building for?
```
4 Years in total. 1 Year for the device
```

## SELinux status?
```
Enforcing
```

## Are you using a prebuilt kernel? If yes, why?
```
No.
```

## Contact
### Telegram username
```
k_y_c_i_i_9_1
```

RisenID, 2024-12-09:
Hey, 
I understand that you may be expecting a quick reply about your application, but trying to speed it up by PMing members of the team is not tolerated. Please feel free to wait for us to review your request, or close the issue if you cant wait.
Thanks

ZorEl212, 2024-12-09:
Hi,
I’m really sorry if it came across that way, I didn’t mean to rush things. I understand and I’m in no hurry, so I’m happy to wait for the review. Sincere apologies!

hipexscape, 2024-12-23:
Please apply in the future when you have gained more knowledge/experience.

closed 2024-12-23 by hipexscape

ZorEl212, 2024-12-23:
@hipexscape can you tell me what is it your looking for regarding skills/knowledge in particular? Thanks