RisingTechOSS-devices/official_devices #12: New maintainer for Nothing Phone 2 (Pong)

samrth012, 2024-12-15:
## Device you want to apply for
```
Nothing Phone 2 (Pong) 
```

## Email address
```
samrth012@gmail.com
```

## Device tree sources

```
https://github.com/samrth012/android_device_nothing_Pong
https://github.com/samrth012/kernel_nothing_sm8475-meteoric
https://github.com/samrth012/vendor_nothing_Pong
```

## Link to the platform where you have released your unofficial build
```
https://t.me/NothingPhone2/411287
https://t.me/NothingPhone2/414057
https://t.me/NothingPhone2/417251
https://t.me/NothingPhone2/497057
```

## Other ROM's you maintain officially or unofficially
```
PixelOS - Unofficial 
```

## Bugs on unofficial build
```
Lawnchair Not working
```

## How long have you been building for?
```
2-2.5 years
```

## SELinux status?
```
Enforcing
```

## Are you using a prebuilt kernel? If yes, why?
```
No.
```

## Contact
### Telegram username
```
samrth012
```

manidweep, 2024-12-16:
a question, i see more than pixel os released by you
namely infinityx and everest os
what about them?

samrth012, 2024-12-16:
> a question, i see more than pixel os released by you namely infinityx and everest os what about them?

For Inifinity x member from the core team had it reserved for official but didn't release any we had discussion about that and he continued it. 
Also for EverestOS dhruvesh from the core team came after the plan of initial release, so he continued further.

manidweep, 2024-12-16:
so you are maintaining them, right?

samrth012, 2024-12-16:
> so you are maintaining them, right?

No as of now, not, they completely took over for the maintenance

samrth012, 2024-12-18:
Any updates sir ?

RisenID, 2024-12-22:
The only issue im seeing here is the presence of vendorsetup.sh, please read the maintainers CoC and requirements listed here
https://github.com/RisingTechOSS-devices/official_devices/blob/main/maintainerreq.md

samrth012, 2024-12-23:
> The only issue im seeing here is the presence of vendorsetup.sh, please read the maintainers CoC and requirements listed here https://github.com/RisingTechOSS-devices/official_devices/blob/main/maintainerreq.md

Sure sir, I have removed it anything else ?

samrth012, 2024-12-26:
Anything else sir ?

> The only issue im seeing here is the presence of vendorsetup.sh, please read the maintainers CoC and requirements listed here https://github.com/RisingTechOSS-devices/official_devices/blob/main/maintainerreq.md

Anything else sir ?

hipexscape, 2024-12-31:
@RisenID

RisenID, 2025-01-02:
> @RisenID

Looks good to me

hipexscape, 2025-01-02:
Accepted

hipexscape, 2025-01-02:
Hey @samrth012,

Please contact me on telegram. My username as my GitHub username.

closed 2025-01-02 by hipexscape