RisingTechOSS-devices/official_devices #14: New maintainer Garnet (Xiaomi Redmi Note 13 Pro 5G/Poco X6 5G)

kibria5, 2024-12-17:
## Device you want to apply for
```
Xiaomi Redmi Note 13 Pro 5G/Poco X6 5G (Garnet)
```

## Email address
```
mdkibria687@gmail.com
```

## Device tree sources
```
https://github.com/kibria5/android_device_xiaomi_garnet.git
https://github.com/kibria5/android_device_xiaomi_garnet-kernel.git
https://github.com/kibria5/android_vendor_xiaomi_garnet.git
```

## Link to the platform where you have released your unofficial build
```
Insert here
```

## Other ROM's you maintain officially or unofficially
```
PixelStar
```

## Bugs on unofficial build
```
None
```

## How long have you been building for?
```
 3 years+
```

## SELinux status?
```
Enforcing.
```

## Are you using a prebuilt kernel? If yes, why?
```
yes. Cause xiaomi release broken kernel source
```

## Contact
### Telegram username
```
@kibria5
```

manidweep, 2024-12-17:
can you tell me why u rebased everything rather than just pick a tree and commit on top of it?
cause i see the commits that u did , were also committed long back by other devs
u did a rebase i get that but why?

kibria5, 2024-12-17:
> can you tell me why u rebased everything rather than just pick a tree and commit on top of it? cause i see the commits that u did , were also committed long back by other devs u did a rebase i get that but why?

I rebased my whole trees and removed some changes and fixes I don't want anyone to get it

manidweep, 2024-12-17:
"I dont want anyone to get it"?
how does this make sense
i just asked you the reason?
why getting rude about it?

kibria5, 2024-12-17:
> "I dont want anyone to get it"? how does this make sense i just asked you the reason? why getting rude about it?

Sorry i didn't say it rudely,I was just trying to explain that I don’t want it shared.

hipexscape, 2024-12-18:
> "I dont want anyone to get it"? how does this make sense i just asked you the reason? why getting rude about it?

Garnet community internal issues.

We should not interfere in that.

donjohanliebert, 2024-12-20:
> > "I dont want anyone to get it"? how does this make sense i just asked you the reason? why getting rude about it?
> 
> Garnet community internal issues.
> 
> We should not interfere in that.

I agree to this.

joshuah345, 2024-12-20:
no issues or concerns on my end 
+1

hipexscape, 2024-12-20:
Looks fine to me 

+1

hipexscape, 2024-12-20:
Hey @kibria5,

Please send a dm to me on telegram 

My username - hipexscape 

Regards,
@hipexscape

closed 2024-12-20 by hipexscape