RisingTechOSS-devices/official_devices #18: New maintainer for Redmi note 8/8T

ZorEl212, 2024-12-28:
## Device you want to apply for
```
Redmi Note 8/8T (ginkgo/willow)
```

## Email address
```
yabsirad212@proton.me
```

## Device tree sources

```
https://github.com/ZorEl212/android_device_xiaomi_ginkgo-retrofit
https://github.com/ZorEl212/android_kernel_xiaomi_ginkgo
https://github.com/ZorEl212/proprietary_vendor_xiaomi_ginkgo
```

## Link to the platform where you have released your unofficial build
```
https://t.me/GinkgoUpdates/3842
```

## Other ROM's you maintain officially or unofficially
```
ArrowOS Extended 
```

## Bugs on unofficial build
```
None
```

## How long have you been building for?
```
4 Years in total. 1 Year for the device
```

## SELinux status?
```
Enforcing
```

## Are you using a prebuilt kernel? If yes, why?
```
No.
```

## Contact
### Telegram username
```
k_y_c_i_i_9_1
```

eun0115, 2024-12-29:
https://github.com/RisingTechOSS-devices/official_devices/issues/9

you are rejected last week, what are your thoughts? 

@hipexscape @RisenID @manidweep

hipexscape, 2024-12-29:
> #9
> 
> you are rejected last week, what are your thoughts?
> 
> @hipexscape @RisenID @manidweep

Please go ahead as I told him to reapply after fixing some device issue.

RisenID, 2025-01-02:
> Please go ahead as I told him to reapply after fixing some device issue.

I would like to wait a while before looking at his work to make sure he has really improved, or if he has just fixed the issues we found for now and thengoing back to those issues later

ZorEl212, 2025-01-02:
Good day @RisenID,

The issues were related to missing authorship on 1 or 2 commits from a while back, which have been fixed. Additionally, the naming convention I previously followed for the device codename (uppercase 'G' in 'Ginkgo') has been updated. Based on recommendation from @hipexscape  to use full lowercase, I have switched to that and will continue with it going forward.

hipexscape, 2025-01-18:
Looks file from my end