RisingTechOSS-devices/official_devices #19: New maintainer for Pixel Fold

austineyoung2000, 2024-12-30:
## Device you want to apply for
``` Pixel Fold
Your device name and codename
``` pixel fold & ( Felix )

## Email address
``` austineyoung2000@gmail.com 

For contact purposes
```

## Device tree sources

```
https://github.com/austineyoung2000/android_device_google_felix
https://android.googlesource.com/device/google/felix-kernels/5.10/
https://github.com/TheMuppets/proprietary_vendor_google_felix
```

## Link to the platform where you have released your unofficial build
```
Insert here
``` 
https://sourceforge.net/p/rising-os-for-pixel-fold/

## Other ROM's you maintain officially or unofficially
```
Insert here
```
Pixel 7, 7 Pro, 7a
Pixel 8, 8 Pro, 8a

## Bugs on unofficial build
```
None
```

## How long have you been building for?
```
Weeks/Months/Years
``` 3+ years

## SELinux status?
```
Enforcing/Permissive.
``` ENFORCING

## Are you using a prebuilt kernel? If yes, why?
```
ITS A PIXEL 
```

## Contact
### Telegram username
``` @EliteDarkKaiser

hipexscape, 2024-12-31:
Atleast fill the form correctly

austineyoung2000, 2024-12-31:
> Atleast fill the form correctly

Done

hipexscape, 2024-12-31:
Looks fine to me 
@RisenID please check

manidweep, 2025-01-02:
approved

closed 2025-01-02 by manidweep