RisingTechOSS-devices/official_devices #21: New maintainer for Poco F4 (munch)

rik-x777, 2025-01-06:
## Device you want to apply for
``` 
POCO F4 (munch)
```

## Email address
```
aritrarik536@protonmail.com
```

## Device tree sources

```
https://github.com/rik-x777/android_device_xiaomi_munch
https://github.com/EmanuelCN/kernel_xiaomi_sm8250
https://codeberg.org/hdzungx/android_vendor_xiaomi_munch
```

## Link to the platform where you have released your unofficial build
```
https://t.me/munch_chat/18387
```

## Other ROM's you maintain officially or unofficially
```
crDroid and Project-PixelStar
```

## Bugs on unofficial build
```
None
```

## How long have you been building for?
```
5 months
```

## SELinux status?
```
Enforcing
```

## Are you using a prebuilt kernel? If yes, why?
```
No.
```

## Contact
### Telegram username
```
rik_x290
```

manidweep, 2025-01-06:
why cherry pick all the old commits ?
and you said 5 months, i dont see your previous work
any comment on that?

rik-x777, 2025-01-07:
https://sourceforge.net/projects/rik-munch/
Here is all my works. Also you can check my group or the poco f4 updates group in telegram to see the posts for all of my builds

manidweep, 2025-01-07:
i'm asking about your git history

rik-x777, 2025-01-07:
I cherry picked mainly from 2 repos. madmax decommonized trees and hdzungx tress

manidweep, 2025-01-07:
and i dont see your previous work except the last week or so
why is that?

rik-x777, 2025-01-07:
Please check the sourceforge for my older works. And If you mean in git, I deleted my previous repos which was based on lineage trees. Many of the commits from madmax's tree was not working out properly so I switched to his trees. 
https://github.com/crdroidandroid/android_device_xiaomi_munch
See this. Its my old tree

RisenID, 2025-01-07:
Do you not maintain your own vendor and kernel? I see some commits to extract script, but its a useless commit if you dont actually use it. Why merge useless code?

rik-x777, 2025-01-07:
I can't extract my vendor as I build in crave where you just give all your trees and build. No way to extract vendor myself. And as for kernel, the N0kernel is excellent for poco f4 so I don't maintain my kernel

hipexscape, 2025-01-07:
> I can't extract my vendor as I build in crave where you just give all your trees and build. No way to extract vendor myself. And as for kernel, the N0kernel is excellent for poco f4 so I don't maintain my kernel. But if needed, I can revert the commits to the extract script

There is still no use of extract utils if you don't have your vendor. It doesn't change anything if you do a commit there cause it's not included on vendor

rik-x777, 2025-01-07:
I have deleted the files. if you want me to revert the commits to for those files, I will do that

hipexscape, 2025-01-07:
> I have deleted the files. if you want me to revert the commits to for those files, I will do that

Deleted ?
Then you should have your own vendor repo perhaps?

rik-x777, 2025-01-07:
Like I said, I don't. And that's why I deleted those as those files are redundant to me. But are you saying that I should revert the commits to those files instead?