RisingTechOSS-devices/official_devices #22: New maintainer for Xiaomi pad 5 (nabu)

hxfuxyy, 2025-01-08:
## Device you want to apply for
Xiaomi Pad 5
Nabu

## Email address
viitiv.mmm@gmail.com

```

## Device tree sources
```
https://github.com/hxfuxyy/android_device_xiaomi_nabu

https://github.com/dev-harsh1998/android_kernel_xiaomi_nabu

https://gitlab.com/dev-harsh1998/android_vendor_xiaomi_nabu

https://github.com/Evolution-X-Devices/hardware_xiaomi

```

## Link to the platform where you have released your unofficial build
```
https://t.me/MiPad_5/435
```

## Other ROM's you maintain officially or unofficially
```
Unofficially- crdroid, pixel project, evoX
```

## Bugs on unofficial build
```
None
```

## How long have you been building for?
```
3-6Months
```

## SELinux status?
```
Enforcing.
```

## Are you using a prebuilt kernel? If yes, why?
```
No.
```

## Contact
### Telegram username
```
@hxfuxyy
```

hipexscape, 2025-01-08:
Sorry but maintaining more than 3 roms is against of our CoC.

hxfuxyy, 2025-01-08:
Okay, then I won’t maintaining this roms

manidweep, 2025-01-08:
if you have been building for a year, why is your github new?

hxfuxyy, 2025-01-08:
> if you have been building for a year, why is your github new?

Because my old account suspended

hxfuxyy, 2025-01-08:
[image]

manidweep, 2025-01-08:
violation of terms?
so you did something in github?

hxfuxyy, 2025-01-08:
> violation of terms? so you did something in github?

Idk really, I’m just forked crave io repo for build roms
I can’t ask it in github support :(

manidweep, 2025-01-08:
can i know your previous git id?

hxfuxyy, 2025-01-08:
> can i know your previous git id?

Sure
“zxcwsurx”

hipexscape, 2025-01-08:
> ![IMG_0368](https://private-user-images.githubusercontent.com/190541853/400973316-757ff1d5-9bcb-4489-87f6-dd45ac9580de.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYzMzEwOTEsIm5iZiI6MTczNjMzMDc5MSwicGF0aCI6Ii8xOTA1NDE4NTMvNDAwOTczMzE2LTc1N2ZmMWQ1LTliY2ItNDQ4OS04N2Y2LWRkNDVhYzk1ODBkZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwOFQxMDA2MzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kOTM1MjA1NTgyOWRiMjBkOTRmYmNjOTE0MDg0MWExNDVkYjBjZTcxZTNjMDQzMGIzZTE4N2E4YmY2OWJlNzdlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.gr3Zh5EgMjdsuEeosEKUD-IpnlUyIRb14VzJrvnKWVY)

Prove it that it's yours only

hxfuxyy, 2025-01-08:
How to prove that?

hxfuxyy, 2025-01-08:
?

hxfuxyy, 2025-01-08:
??

hxfuxyy, 2025-01-09:
?

closed 2025-01-11 by hxfuxyy