RisingTechOSS-devices/official_devices #26: New maintainer for POCO X3/NFC(surya)

shoya0x00, 2025-01-11:
## Device you want to apply for
```
POCO X3/NFC(surya)
```

## Email address
```
shoya0x00@gmail.com
```

## Device tree sources

```
https://github.com/surya-aosp/device_xiaomi_surya
https://github.com/surya-aosp/kernel_xiaomi_surya
https://gitlab.com/apaosha/vendor_xiaomi_surya
```

## Link to the platform where you have released your unofficial build
```
https://t.me/braindeadci/21
```

## Other ROM's you maintain officially or unofficially
```
crDroid
Pixelage
```

## Bugs on unofficial build
```
None
```

## How long have you been building for?
```
around 1 years
```

## SELinux status?
```
Enforcing
```

## Are you using a prebuilt kernel? If yes, why?
```
No.
```

## Contact
### Telegram username
```
braindead88
```

donjohanliebert, 2025-01-14:
ahh im not assigned here XD

hipexscape, 2025-01-14:
> ahh im not assigned here XD

Feel free to review

shoya0x00, 2025-01-14:
> ahh im not assigned here XD

Sir go easy on me. Me noob 🙂🙂

donjohanliebert, 2025-01-14:
> > ahh im not assigned here XD
> 
> Sir go easy on me. Me noob 🙂🙂

pm on tg @honokascape asap thanks

hipexscape, 2025-01-14:
Approved

hipexscape, 2025-01-14:
Hey @shoya0x00,

Please send a dm to me on telegram

My username - hipexscape

Regards,
@hipexscape

closed 2025-01-14 by hipexscape