RisingTechOSS-devices/official_devices #27: New maintainer For Moto G54 5G (cancunf)

ussr1674, 2025-01-11:
## Device you want to apply for
```
Moto G54 5G (cancunf)
```

## Email address
```
ussr1674@gmail.com
```

## Device tree sources
```
https://github.com/ussr1674/device_motorola_cancunf
https://github.com/ussr1674/vendor_motorola_cancunf
https://github.com/ussr1674/device_motorola_cancunf-kernel
```

## Link to the platform where you have released your unofficial build
```
https://t.me/motorolag54official/376151
```

## Other ROM's you maintain officially or unofficially
```
crDroid
Project Matrixx
```

## Bugs on unofficial build
```
None
```

## How long have you been building for?
```
Months(Almost Year)
```

## SELinux status?
```
Enforcing
```

## Are you using a prebuilt kernel? If yes, why?
```
No. Using DH kernel Because It is Stable kernel For Our device Which Has Fixed Kernel Related Issues
```

## Contact
### Telegram username
```
https://t.me/ussr_1674
```

hipexscape, 2025-01-12:
Soo less contribution on trees, why ?

ussr1674, 2025-01-12:
> Soo less contribution on trees, why ?

Cause In Our G54 Community We Use Sarthak Sir Trees As It is stable And Everyone In Our Community Do Use It And Like Previous RisingOS Maintainer Ayan Did For Moto G54 5G.

manidweep, 2025-01-12:
he has had more experience and had his own trees based on sarthak
but as i heard from ayan, you didnt even have a tree of your own until you wanted to apply for maintainership
you say you had a year of experience, but where does that even showup in your git?

ussr1674, 2025-01-12:
> he has had more experience and had his own trees based on sarthak but as i heard from ayan, you didnt even have a tree of your own until you wanted to apply for maintainership you say you had a year of experience, but where does that even showup in your git?

I Do Changes In My Local Server As I Was Maintaining For Other Roms crdroid so I decided to rebase To avoid messup

manidweep, 2025-01-12:
so you didnt bother to push to git and update your trees before?

ussr1674, 2025-01-12:
> so you didnt bother to push to git and update your trees before?

Nah I did but for crdroid I deleted previous and rebased currently for RisingOS as Ayan left maintainership then I decided to do for RisingOS.

hipexscape, 2025-01-12:
> > so you didnt bother to push to git and update your trees before?
> 
> Nah I did but for crdroid I deleted previous and rebased currently for RisingOS as Ayan left maintainership then I decided to do for RisingOS.

I don't see contribution on your other tree either https://github.com/ussr1674/device_motorola_devonf/commits/fourteen/

ussr1674, 2025-01-12:
> I don't see contribution on your other tree either https://github.com/ussr1674/device_motorola_devonf/commits/fourteen/

Thats Not My Primary Device Actually There Community is ded No Roms So I Decided To do Project Matrixx For Them.

manidweep, 2025-01-12:
> Thats Not My Primary Device Actually There Community is ded No Roms So I Decided To do Project Matrixx For Them.

do you own that device?

ussr1674, 2025-01-12:
> do you own that device?

No

ussr1674, 2025-01-12:
> do you own that device?

But they have same soc group considering us MT6855 Also Sarthak Sir Were Helping There Community. so I decided to do too.

hipexscape, 2025-01-12:
> No

Should we consider them as blind building ?

ussr1674, 2025-01-12:
> Should we consider them as blind building ?

Nah. They Belong to same soc grp MT6855 Like Moto G54/64 . So Did

ussr1674, 2025-01-12:
What's Your Opinion Sir Now?

sarthakroy2002, 2025-01-12:
> > > so you didnt bother to push to git and update your trees before?
> > 
> > 
> > Nah I did but for crdroid I deleted previous and rebased currently for RisingOS as Ayan left maintainership then I decided to do for RisingOS.
> 
> I don't see contribution on your other tree either https://github.com/ussr1674/device_motorola_devonf/commits/fourteen/
> 
> Should i guess you just do bringup ?

I did the bringup? Any problems on it? It was 95% similar device, so i did a minimal bringup so that others can atleast follow cancunf commits and uprev devonf to newer android versions.

sarthakroy2002, 2025-01-12:
> Should we consider them as blind building ?

Then you should stop maintaining common trees which many devices use?

hipexscape, 2025-01-12:
Sarthak on fire......

Approved from my end

Sureshgorai25, 2025-01-12:
[image]
I hope you guys will fix this.
The codename of the device the cancunf not cancuf

hipexscape, 2025-01-12:
> ![Screenshot_20250112-153957_Cromite](https://private-user-images.githubusercontent.com/93507842/402347645-a2bf8e6f-6a99-4182-bcb0-b5b41e91ea3d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY2ODUwMDUsIm5iZiI6MTczNjY4NDcwNSwicGF0aCI6Ii85MzUwNzg0Mi80MDIzNDc2NDUtYTJiZjhlNmYtNmE5OS00MTgyLWJjYjAtYjViNDFlOTFlYTNkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEyVDEyMjUwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ5OWVmODM0YjdhYWFlNzZhMmU4NjRjY2QxOThiNjJjOGYwODA0YzlkZjYyZjNhNzRkNjFkY2ZmNWZjMTExZGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.QWkEKAn_HDYA-w66q5A2Qu2Uj1nSfSi6uu0bCHkLJuY) I hope you guys will fix this. The codename of the device is cancunf not cancuf

Surely sir

ussr1674, 2025-01-12:
@manidweep What's Your Opinion Now?

manidweep, 2025-01-12:
just woke up bud, chill out

> @manidweep What's Your Opinion Now?

ussr1674, 2025-01-12:
> just woke up bud, chill out
> 
> > @manidweep What's Your Opinion Now?

We are gonna sleep now

Sureshgorai25, 2025-01-12:
😂

ussr1674, 2025-01-13:
> just woke up bud, chill out
> 
> > @manidweep What's Your Opinion Now?

Sorry for inconvenience

manidweep, 2025-01-13:
why are you in such a hurry @ussr1674 ?
i cannot be busy or what?

manidweep, 2025-01-13:
> > why are you in such a hurry @ussr1674 ? i cannot be busy or what?
> 
> Not In Hurry. Just Waiting

the why make fun of like "Hmm he is busy whole day or what 😂😂"

manidweep, 2025-01-13:
laughing emoji, can see

ussr1674, 2025-01-13:
> laughing emoji, can see

My Bad

manidweep, 2025-01-13:
if you have no patience, i cannot help it fr

ussr1674, 2025-01-13:
> if you have no patience, i cannot help it fr

I Will wait.

Sureshgorai25, 2025-01-13:
Ohh

manidweep, 2025-01-13:
> Ohh

who are you?

Sureshgorai25, 2025-01-13:
> > Ohh
> 
> who are you?

A g54 user, why do u ask tho?

closed 2025-02-22 by ussr1674