RisingTechOSS-devices/official_devices #29: New maintainer for Pixel Fold Tangorpro

austineyoung2000, 2025-01-21:
## Device you want to apply for
```
Pixel Fold
```
Your device name and codename
```
tangorpro
```

## Email address
```
austineyoung2000@gmail.com 
```

## Device tree sources

```
https://github.com/austineyoung2000/android_device_google_tangorpro
https://github.com/austineyoung2000/proprietary_vendor_google_tangorpro
https://android.googlesource.com/device/google/tangorpro-kernels/5.10/
```

## Link to the platform where you have released your unofficial build
```
https://xdaforums.com/t/rom-unofficial-beta-community-rising-os-6-1-gapps-nov-spl-a15.4708507/#post-89898066
```

## Other ROM's you maintain officially or unofficially
```
Pixel 8 series ( 8, 8 Pro, 8a )
Pixel 7 Series ( 7, 7 Pro, 7a )
Pixel Fold
```

## Bugs on unofficial build
```
Prior too todays sync and tree push I had a pixel launcher crash and camera ,  and blur effects where broken.. todays tree push fixed these issues
```

## How long have you been building for?
```
3 Years
```

## SELinux status?
```
Enforcing
```

## Are you using a prebuilt kernel? If yes, why?
```
Yes, as all pixels have this. Their a few inline kernels but their nothing like stock 
```

## Contact
### Telegram username
```
EliteDarkKaiser
```

austineyoung2000, 2025-01-21:
Fixed name

manidweep, 2025-01-26:
approved

closed 2025-01-26 by manidweep