RisingTechOSS-devices/official_devices #30: New maintainer

kjdev16, 2025-01-26:
## Device you want to apply for
```
Moto G73 5g
devonf
Your device name and codename
```

## Email address
```
kjdev00016@gmail.com
For contact purposes
```

## Device tree sources

```
https://github.com/Adarsh0-s/device_motorola_devonf/tree/fifteen-rising
https://github.com/Adarsh0-s/android_device_motorola_devonf-kernel
https://github.com/Adarsh0-s/android_vendor_motorola_devonf
```

## Link to the platform where you have released your unofficial build
```
https://t.me/MotoG73Official/28910
https://drive.google.com/drive/folders/1YHXiGuA-s-XjT4_UEJUfq1kv_z0A5fuH?usp=drive_link
Insert here
```

## Other ROM's you maintain officially or unofficially
```
NA
```

## Bugs on unofficial build
```
Double Tap to Wake
```

## How long have you been building for?
```
2 weeks
```

## SELinux status?
```
Enforcing.
```

## Are you using a prebuilt kernel? If yes, why?
```
No.
```

## Contact
### Telegram username
```
@genzimplore
Your Telegram username
```

hipexscape, 2025-01-26:
Not approved 

Not a single contribution on your tree

joshuah345, 2025-01-26:
https://github.com/Adarsh0-s/device_motorola_devonf/commit/dda1e751a8877578dd8212c0600ba2ac794787d0

Can you explain why you squashed serveral commits into one (imporoperly) and didn't properly provide authorship here? Just mentioning the names of the original authors is not the way to do it either.

kjdev16, 2025-01-26:
Actually the commits of device trees I used is made by Adarsh, not me.

joshuah345, 2025-01-26:
As a maintainer, there's still an expectation to know what's in your trees and be able to fix issues if and when they occur
You're also expected to have basic knowledge of git to track such changes 

it may not be your tree but you will be responsible for it

kjdev16, 2025-01-26:
Thank you @joshuah345 for explaining. Actually I am kinda new to this AOSP projects as you might see from my GitHub. I'll revoke my current application.

Also, after how long I would be able to apply again for a maintainer?

manidweep, 2025-01-26:
3 monthss

> Thank you [@joshuah345](https://github.com/joshuah345) for explaining. Actually I am kinda new to this AOSP projects as you might see from my GitHub. I'll revoke my current application.
> 
> Also, after how long I would be able to apply again for a maintainer?

closed 2025-01-26 by manidweep

RisenID, 2025-01-26:
> 3 monthss

To add to this, even after the period has passed, repeated applications made when your work isnt can will lead to harsher outcomes.